SSL-A-Team/firmware #49: update-alternatives fails in udev_setup.bash when alternative is already set?

joe-spall, 2024-01-15:
On WSL2 if running udev_setup.bash after update-alternatives has been set for usbip, the line  
`update-alternatives --install /usr/local/bin/usbip usbip /usr/lib/linux-tools/*-generic/usbip 20` fails with
`update-alternatives: priority must be an integer`

This causes the rest of the setup script to not be run. This error also occurs if the line is ran from the command line. I am speculating its because the priority is already set? 20 is clearly an integer.

guyfleeman, 2024-01-15:
Oh man okay. Uh I just ran this on my WSL2 setup so maybe we can diff errors or something

joe-spall, 2024-01-17:
I had followed the WSL2 instructions manually last year but wanted to try to new script so that's why I am guessing it's because it is already set

closed 2025-06-04 by joe-spall