SSL-A-Team/firmware #50: udevadm control --reload in udev_setup.bash failed if already running?

joe-spall, 2024-01-15:
`udevadm control --reload` failed if already running. Got the error of 
`Failed to send reload request: No such file or directory`

Resolved by running 
`sudo service udev restart`
and the `udevadm` commands needed

Maybe catch on this error and echo with suggestion of running above?

guyfleeman, 2024-01-15:
I think this might mean it's not running? Any reason we can't just run the restart command every time? This is a one time setup, so if it causes some minor device blip it'd be a one off and reload might do that anyway?

guyfleeman, 2024-01-15:
Otherwise we can conditionally check for the failure and run restart followed by a second reload.

Is this WSL2 Win 11 or Ubuntu native?

joe-spall, 2024-01-17:
Probably just running restart is fine. On WSL2 Ubuntu